redballoonsecurity/ofrak #3: Nonfree license

TechnologyClassroom, 2022-08-11:
I saw your tool mentioned in [Wired](https://www.wired.com/story/ofrak-iot-reverse-engineering-tool/).  The article says OFRAK is open source, but the [license](https://github.com/redballoonsecurity/ofrak/blob/master/LICENSE) says it is only for non-commercial usage.

Non-commercial (NC) clauses are nonfree so OFRAK is not free software according to the [FSF](https://www.gnu.org/philosophy/free-sw.html#selling) or open source according to the [OSI](https://opensource.org/osd#free-redistribution).  I would recommend removing the non-commercial clause from your license.

Querela, 2022-08-11:
Isn't that more an issue of the WIRED article being not exact in its wording? In laymen's terms it is "open" source since you can look at the source... That it is not completely open/free by definition is another point. The "O" in OFRAK might be confusing, though. 
And as the authors provided their own license we should respect that. I can understand wanting some remuneration for your work when it is used commercially. (Publicly funded projects may be different but essentially anyone can use it ...)

TechnologyClassroom, 2022-08-11:
The O in OFRAK probably led the WIRED article to assume it was open source, but you are correct in that OFRAK has not called itself open source.  Regardless, releasing the code with a free software and open source license would foster a community effort to improve on the project.  NC is a dead-end.

EdwardLarson, 2022-08-11:
OFRAK is source-available, but not open source. As the License section of the README explains, the code in this repo comes with the OFRAK Community License, which is intended for educational uses, personal development, or just having fun. Users interested in using OFRAK for commercial purposes can request the Pro License at [ofrak.com/license](https://ofrak.com/license/) , which for a limited period is available for a free 6-month trial.

While Ang did consider open sourcing FRAK, as the Wired article notes, we believe that now, 10 years later, this source-available model for OFRAK strikes the right balance between, on the one hand, making OFRAK available to the community and, on the other hand, protecting our interests as an early-stage company. We also think this model best supports the ongoing development of OFRAK.

We intend to add a similar note in the FAQ we are planning to put together in the near future.

closed 2022-08-11 by EdwardLarson