redballoonsecurity/ofrak #51: Add ZStandard components

whyitfor, 2022-10-04:
**What is the use case for the feature?**
ZStandard is a modern compression format known for its speed and high compression ratios.

- <https://github.com/facebook/zstd>
- <http://facebook.github.io/zstd/>

It has a command line tool that can be installed with:

``` bash
# Ubuntu / Debian / OFRAK Docker
sudo apt install zstd
```

Or:

``` bash
# macOS
brew install zstd
```

The packer and unpacker components would probably end up being very similar to the [LZO components](https://github.com/redballoonsecurity/ofrak/blob/master/ofrak_components/ofrak_components/lzo.py) in structure.

There are some example ZStandard-compressed files [here](https://github.com/indygreg/python-build-standalone/releases), but they may not make for great test files because they're fairly large, so it is probably worth finding better (or smaller) ones...

**Does the feature contain any proprietary information about another company's intellectual property?**
No, the format is open-source https://github.com/facebook/zstd/blob/dev/LICENSE.

**How would you implement this feature?**
See above.

**Are there any (reasonable) alternative approaches?**
N/A.

**Are you interested in implementing it yourself?**
No -- this is a great first contributor issue!

thejoelpatrol, 2022-10-07:
I will take a stab at this enhancement for the super cool and good ofrak project next week.

whyitfor, 2022-10-21:
Closed by #72.

closed 2022-10-21 by whyitfor