redballoonsecurity/ofrak #53: Use asyncio interface for subprocess calls

whyitfor, 2022-10-04:
Many components use `subprocess` to make calls to external programs via the command line. These are currently blocking, non-asynchronous calls.

The [`asyncio.create_subprocess_exec` and `asyncio.create_subprocess_shell` functions](https://docs.python.org/3/library/asyncio-subprocess.html) represent an `asyncio`-compatible API for making subprocess calls that don't block the event loop. We should replace existing `subprocess` calls in OFRAK with async calls via this API.

**Which files would be affected?**
Any file that uses a `subprocess` call.

**Does the proposed maintenance include non-doc string _functional_ changes to the Python code?**
Yes (see above)

**Are you interested in implementing it yourself?**
No -- this is a great first contributor issue!

marczalik, 2023-02-21:
@rbs-jacob Should this change be made for `subprocess` calls inside of fixtures and test cases too?

rbs-jacob, 2023-02-21:
> @rbs-jacob Should this change be made for `subprocess` calls inside of fixtures and test cases too?

That seems less important to me. The main issue is that the `subprocess` calls currently block the event loop to unpack a filesystem, for example. This lag directly affects a user by fully blocking OFRAK concurrency for an inherently asynchronous operation, whereas the async concurrency is less meaningful in the tests. This is because the tests are either run fully serially, or explicitly concurrently using multiprocessing via pytest. 

Does that clarify?

closed 2023-02-28 by rbs-jacob