redballoonsecurity/ofrak #54: Use pigz instead of gzip for Gzip components

whyitfor, 2022-10-04:
`pigz` is a parallel version of `gzip` also written by Mark Adler (the co-creator of gzip/zlib). It acts as a sort of drop-in replacement for the `gzip` command line utility, but it's parallelized and much faster.

We should replace uses of `gzip` in OFRAK with `pigz`.

**Which files would be affected?**
https://github.com/redballoonsecurity/ofrak/blob/master/ofrak_components/ofrak_components/zip.py

**Does the proposed maintenance include non-doc string _functional_ changes to the Python code?**
Yes -- changing the command used and the Dockerstub file.

**Are you interested in implementing it yourself?**
No. This is a great first contributor issue!

rbs-jacob, 2022-10-05:
Hi @The-Debarghya, I've assigned you this issue. Repeating my comment on the other issue you requested to work on:

> To make this change, you will likely have to change some `subprocess.run` calls in OFRAK components, as well as changing installed dependencies in one of the `Dockerstub` files.
> 
> Make sure to look at the [contributing guide](https://github.com/redballoonsecurity/ofrak/blob/master/docs/contributor-guide/getting-started.md). Thanks for your interest in helping out!

alexreuter, 2022-10-20:
Hi there, I would like to take a look at this! Thanks