redballoonsecurity/ofrak #55: Use official version of 7zip on Linux instead of unsupported p7zip

whyitfor, 2022-10-04:
See: <https://news.ycombinator.com/item?id=31846317>

**Which files would be affected?**
https://github.com/redballoonsecurity/ofrak/blob/master/ofrak_components/ofrak_components/p7zip.py

**Does the proposed maintenance include non-doc string _functional_ changes to the Python code?**
Yes.

**Are you interested in implementing it yourself?**
No -- this is a great first issue!

The-Debarghya, 2022-10-05:
Hey, can you assign this to me?

rbs-jacob, 2022-10-05:
Hi @The-Debarghya, I've assigned you this issue. To make this change, you will likely have to change some `subprocess.run` calls in OFRAK components, as well as changing installed dependencies in one of the `Dockerstub` files.

Make sure to look at the [contributing guide](https://github.com/redballoonsecurity/ofrak/blob/master/docs/contributor-guide/getting-started.md). Thanks for your interest in helping out!

The-Debarghya, 2022-10-07:
I'm struggling to figure out where to change the `subprocess.run` calls in the OFRAK components...can you please help me out.

whyitfor, 2022-10-07:
@The-Debarghya,

`7z` is run in two places:
- https://github.com/redballoonsecurity/ofrak/blob/master/ofrak_components/ofrak_components/p7zip.py#L48
- https://github.com/redballoonsecurity/ofrak/blob/master/ofrak_components/ofrak_components/p7zip.py#L68

Variables using the name `p7zip` should be updated to use 7zip.

Instructions/code that installs `p7zip` should also be updated to install 7zip:
- https://github.com/redballoonsecurity/ofrak/blob/master/ofrak_components/Dockerstub#L12
- https://github.com/redballoonsecurity/ofrak/blob/master/docs/environment-setup.md?plain=1#L184

You can test the changes by running the following test: https://github.com/redballoonsecurity/ofrak/blob/master/ofrak_components/ofrak_components_test/test_p7zip_component.py. The test file might need to be updated as well.

Hope this helps!

The-Debarghya, 2022-10-13:
p7zip variables can't be renamed to 7zip because python doesn't permit variable name starting with an integer, so should I name it like this: _7zip?
and for the filenames...I'm renaming them to 7zip.

whyitfor, 2022-10-13:
Either `sevenzip` or `seven_zip` probably make sense (based on https://peps.python.org/pep-0008/#function-and-variable-names).

Regarding the filenames that sounds fine.

closed 2023-02-06 by whyitfor