redballoonsecurity/ofrak #56: Add support for El Torito ISO images

whyitfor, 2022-10-04:
Many ISOs from the Internet are El Torito images, which are [currently not supported by the OFRAK ISO unpacker](https://gitlab.com/redballoonsecurity/ofrak/-/blob/master/ofrak_core/ofrak_components/iso9660.py#L190).

Often the El Torito images will unpack successfully, but there are no tests for them, and it is not clear what more would need to be done to support them.

**What is the use case for the feature?**
Support unpacking and modifying El Torito images.

**Does the feature contain any proprietary information about another company's intellectual property?**
No.

**How would you implement this feature?**
Probably by extending current ISO support.

**Are there any (reasonable) alternative approaches?**

**Are you interested in implementing it yourself?**
No. This is a great first contributor issue!

Tanmoy741127, 2022-10-06:
Hi can you assign this to me ?

whyitfor, 2022-10-06:
Hi @Tanmoy741127, I've assigned you this issue.

Make sure to look at the [contributing guide](https://github.com/redballoonsecurity/ofrak/blob/master/docs/contributor-guide/getting-started.md). Thanks for your interest in helping out!

rbs-jacob, 2024-03-14:
Fixed by #394.

closed 2024-03-14 by rbs-jacob