redballoonsecurity/ofrak #60: make FilesystemRoot.initialize_from_disk() fail when given an incorrect path, instead of not initializing anything

whyitfor, 2022-10-04:
**What is the problem? (Here is where you provide a complete Traceback.)**
The current code is essentially:
```
for root, dirs, files in os.walk(root_path):
    <something>
```
Which does nothing when root_path doesn't exist.

It also does nothing when the path given corresponds to a file and not a directory.

**Please provide some information about your environment.**
N/A

**If you've discovered it, what is the root cause of the problem?**
See above.

**How often does the issue happen?**
Whenever an incorrect path is passed to `FilesystemRoot.initialize_from_disk`.

**Are you interested in implementing it yourself?**
No. This is a good first contributor issue!

galbwe, 2022-10-05:
Hi, can I work on this issue?

jstrieb, 2022-10-05:
Hi @galbwe, go for it! Be sure to take a look at the [contributing guide](https://ofrak.com/docs/contributor-guide/getting-started.html).

Feel free to follow up with any questions in this thread.

PR #64 (2022-10-05): FilesystemRoot.initialize_from_disk path validation
**Link to Related Issue(s)**
Closes #60 

**Please describe the changes in your request.**
Adds validation of the path parameter passed to `FilesystemRoot.initialize_from_disk`. Will raise a `FileNotFoundError` if the path is not found or is to a file instead of a directory.

**Anyone you think should look at this, specifically?**
@whyitfor since they opened the issue
closed 2022-10-05 by EdwardLarson