redballoonsecurity/ofrak #76: Run OFRAK on MacBook with M1 chip

whyitfor, 2022-10-18:
**What is the use case for the feature?**
I would like to be able to run OFRAK on my MacBook with the M1 chip.

**Does the feature contain any proprietary information about another company's intellectual property?**
No.

**How would you implement this feature?**
Ensure that Docker image can be built and run on this platform. Additionally, ensure that the Python install of OFRAK on the target platform works.

marczalik, 2023-02-03:
Non-comprehensive fixes and issues that I've run into while using OFRAK on an M1 Mac:

- Need to add `apt-get cmake` to ofrak_core's Dockerstub.
- Running Docker containers as ARMv8 led to issues with Toolchain. LLVM unsupported and using GNU led to insufficient space issues during patch making.
- In order to run Docker as AMD64, need to:
     - Add `FROM --platform=linux/amd64` in `build_image.py` where the Dockerfile parts are specified.
     - Set `ARG TARGETARCH="amd64"` in `ofrak_patch_maker`'s `Dockerstub`.
- Segfaulting when attempting to build Docker images with any Python > 3.7.

whyitfor, 2023-02-12:
On M1 I ran into issue installing orjson, which wanted rust to build from source a dependency. Upgrading pip `python3 -m pip install --upgrade pip` to 23.0 resolved this issue.

Running `pip install ofrak` also ran into issues with `xattr==0.9.7` (because of `cffi`) -- updating to `xattr=0.10.1` resolves that issue (this comes with #217).

With `pip install ofrak`, installing `lief` also ran into issues (cmake was not installed), since there is not a 3.8 wheel available.

whyitfor, 2023-02-12:
I also needed to `brew install libmagic`.

whyitfor, 2025-09-30:
Marking this as closed 
- Core OFRAK runs on MacBook M1 currently on https://github.com/redballoonsecurity/ofrak/commit/5c0ff947768e106f70ede0a10c8b73b7d626155a.
- ofrak 3.3.0 will also install correctly from PyPI (see https://pypi.org/project/ofrak/3.3.0rc14/).
- CI testing of core package is also now happening in CI/CD: https://github.com/redballoonsecurity/ofrak/blob/master/.github/workflows/build-and-test-distributions.yml