redballoonsecurity/ofrak #78: Elf.get_symbol_section fails when more than one symbol section exists

whyitfor, 2022-10-20:
**What is the problem? (Here is where you provide a complete Traceback.)**
`Elf.get_symbol_section` fails when the target elf has more than one symbol section.

**Please provide some information about your environment.**
The following test (using the `elf_executable_file` fixture) reproduces the issue

```
import pytest

from ofrak import OFRAKContext, Resource
from ofrak.core import Elf, ElfSymbolSection, ElfSymbolBinding, ElfSymbolType, \
    ElfSymbolVisibility
from ofrak_type.range import Range

@pytest.fixture
async def elf_resource(elf_executable_file: str, ofrak_context: OFRAKContext) -> Resource:
    return await ofrak_context.create_root_resource_from_file(elf_executable_file)

async def test_elf_view(elf_resource: Resource):
    await elf_resource.unpack()
    elf = await elf_resource.view_as(Elf)
    for elf_section_header in await elf.get_section_headers():
        file_range = elf_section_header.get_file_range()
        assert isinstance(file_range, Range)

    symbol_section = await elf.get_symbol_section()
    assert isinstance(symbol_section, ElfSymbolSection)
    for symbol in await symbol_section.get_symbols():
        assert isinstance(symbol.get_binding(), ElfSymbolBinding)
        assert isinstance(symbol.get_type(), ElfSymbolType)
        assert isinstance(symbol.get_visibility(), ElfSymbolVisibility)
        symbol_section_index = symbol.get_section_index()
        if symbol_section_index is not None:
            assert isinstance(symbol_section_index, int)
```

**If you've discovered it, what is the root cause of the problem?**
The resulting binary has an `ElfSymbolSection` as well as an `ElfDynaSymbolSection`.

**How often does the issue happen?**
N/A.

**What are the steps to reproduce the issue?**
Run the above test in the OFRAK container.

**How would you implement this fix?**
TBD.

**Are there any (reasonable) alternative approaches?**

**Are you interested in implementing it yourself?**

EdwardLarson, 2022-10-20:
I think this is actually a problem that `ElfDynaSymbolSection` is a subclass of `ElfSymbolSection`. At first that seems like a reasonable inheritance, but according to [this](https://docs.oracle.com/cd/E19455-01/806-3773/elf-2/index.html), there should only ever be at most one symbol section, and the dynamic symbol section is something else:

> 

Name | Value |  
-- | -- | --
SHT_SYMTAB | 2 | Is a complete symbol table, usually for link editing. This table can also be used for dynamic linking; however, it can contain many unnecessary symbols. Note: Only one section of this type is allowed in a file
-- | -- | --
SHT_DYNSYM | 11 | Is a symbol table with a minimal set of symbols for dynamic linking.  Note: Only one section of this type is allowed in a file
-- | -- | --

whyitfor, 2022-10-21:
Which one should `Elf.get_symbol_section` return? Which section does [ElfRelocateSymbolsModifier](https://ofrak.com/docs/reference/ofrak/core/elf/modifier.html#ofrak.core.elf.modifier.ElfRelocateSymbolsModifier) intend to target?

EdwardLarson, 2022-10-21:
> Which one should `Elf.get_symbol_section` return?

Definitely the `SHT_SYMTAB`

>Which section does [ElfRelocateSymbolsModifier](https://ofrak.com/docs/reference/ofrak/core/elf/modifier.html#ofrak.core.elf.modifier.ElfRelocateSymbolsModifier) intend to target?

I think that should target both. I'm not too familiar with the SHT_DYNSYM though. It looks like it should modify both as they are both symbol tables, and if a symbol in one is updated it should be synchronized with the other.