redballoonsecurity/ofrak #83: BuildKit requirement is not clear

MatthewTingum, 2022-10-22:
**What is the problem? (Here is where you provide a complete Traceback.)**
25e7e9a9ceab61281cfb933e54e32e905bcf6a36 introduced a change that breaks CMake installation where BuildKit is not enabled.

```sh
$ python3 build_image.py --config ofrak-ghidra.yml --base --finish
...
[SNIP]
...
Step 23/45 : RUN cd /tmp &&     git clone https://github.com/rbs-forks/keystone.git &&     cd keystone &&     git checkout 2021.09.01 &&     ./install_keystone.sh &&     cd /tmp/keystone/bindings/python && python setup.py install &&     cd /tmp &&     rm -r keystone
 ---> Running in c4aa3830570a
Cloning into 'keystone'...
Note: switching to '2021.09.01'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -c with the switch command. Example:

  git switch -c <new-branch-name>

Or undo this operation with:

  git switch -

Turn off this advice by setting config variable advice.detachedHead to false

HEAD is now at f67afb9 Merge remote-tracking branch 'ks_master/master' into upstream_pull_9.2021
../make-share.sh: 21: cmake: not found
./install_keystone.sh: line 6: cmake: command not found
make: *** No targets specified and no makefile found.  Stop.
make: *** No rule to make target 'install'.  Stop.
cp: cannot stat '/usr/local/lib/x86_64-linux-gnu/*': No such file or directory
rm -rf ./build src/
rm -rf prebuilt/win64/keystone.dll
rm -rf prebuilt/win32/keystone.dll
if test -n ""; then \
    python setup.py build -b ./build install --root=""; \
else \
    python setup.py build -b ./build install; \
fi
running build
Building C++ extensions
make[1]: Entering directory '/tmp/keystone/bindings/python'
rm -rf ./build src/ dist/ README
rm -f keystone/*.so
rm -rf prebuilt/win64/keystone.dll
rm -rf prebuilt/win32/keystone.dll
make[1]: Leaving directory '/tmp/keystone/bindings/python'
/tmp/keystone/bindings/python/../../CMakeLists.txt -> /tmp/keystone/bindings/python/src/CMakeLists.txt
/tmp/keystone/bindings/python/../../CMakeUninstall.in -> /tmp/keystone/bindings/python/src/CMakeUninstall.in
/tmp/keystone/bindings/python/../../CMakeLists.txt -> /tmp/keystone/bindings/python/src/CMakeLists.txt
/tmp/keystone/bindings/python/../../LICENSE-COM.TXT -> /tmp/keystone/bindings/python/src/LICENSE-COM.TXT
/tmp/keystone/bindings/python/../../AUTHORS.TXT -> /tmp/keystone/bindings/python/src/AUTHORS.TXT
/tmp/keystone/bindings/python/../../SPONSORS.TXT -> /tmp/keystone/bindings/python/src/SPONSORS.TXT
/tmp/keystone/bindings/python/../../CREDITS.TXT -> /tmp/keystone/bindings/python/src/CREDITS.TXT
/tmp/keystone/bindings/python/../../COPYING -> /tmp/keystone/bindings/python/src/COPYING
/tmp/keystone/bindings/python/../../LICENSE-COM.TXT -> /tmp/keystone/bindings/python/src/LICENSE-COM.TXT
/tmp/keystone/bindings/python/../../EXCEPTIONS-CLIENT -> /tmp/keystone/bindings/python/src/EXCEPTIONS-CLIENT
/tmp/keystone/bindings/python/../../README.md -> /tmp/keystone/bindings/python/src/README.md
/tmp/keystone/bindings/python/../../RELEASE_NOTES -> /tmp/keystone/bindings/python/src/RELEASE_NOTES
/tmp/keystone/bindings/python/../../ChangeLog -> /tmp/keystone/bindings/python/src/ChangeLog
/tmp/keystone/bindings/python/../../SPONSORS.TXT -> /tmp/keystone/bindings/python/src/SPONSORS.TXT
/tmp/keystone/bindings/python/../../pkg-config.pc.cmake -> /tmp/keystone/bindings/python/src/pkg-config.pc.cmake
/tmp/keystone/bindings/python/../../make-share.sh -> /tmp/keystone/bindings/python/src/make-share.sh
/tmp/keystone/bindings/python/../../install_keystone.sh -> /tmp/keystone/bindings/python/src/install_keystone.sh
/tmp/keystone/bindings/python/../../make-common.sh -> /tmp/keystone/bindings/python/src/make-common.sh
/tmp/keystone/bindings/python/../../make-afl.sh -> /tmp/keystone/bindings/python/src/make-afl.sh
/tmp/keystone/bindings/python/../../make-lib.sh -> /tmp/keystone/bindings/python/src/make-lib.sh
[... 67 lines cut ...]
`DOCKER_BUILDKIT=1` must be exported or buildkit must be enabled.

**How often does the issue happen?**
Every time I attempt to build the docker image without buildkit.

**What are the steps to reproduce the issue?**
```sh
$ python3 build_image.py --config ofrak-ghidra.yml --base --finish
```

**How would you implement this fix?**
Update docs to replace:
```sh
python3 build_image.py --config ofrak-ghidra.yml --base --finish
```
with:
```sh
DOCKER_BUILDKIT=1 python3 build_image.py --config ofrak-ghidra.yml --base --finish
```

**Are there any (reasonable) alternative approaches?**
Detect target architecture without the use of buildkit.

**Are you interested in implementing it yourself?**
Yes, after discussing solutions.

MatthewTingum, 2022-10-23:
To further clarify, the `TARGETARCH` variable is not defined without BuildKit.

rbs-jacob, 2022-10-23:
Hey @MatthewTingum, thanks for raising this! One (admittedly brittle) solution is to set `ARG TARGETARCH=amd64` so that there is a default argument in the event that BuildKit is not available. 

In general, the change you cite has messed with the Docker build caching in the CI pipeline, so we're looking into improving that, and can keep this in mind as well.

MatthewTingum, 2022-10-23:
That's an option, but then won't AARCH64 users find themselves with this same issue? Seems like it's just shifting the problem to different users.

MatthewTingum, 2022-10-23:
My knowledge of Docker and BuildKit is surface level. Are Python scripts the only places Dockerfiles are *intended* to be generated and built in this repo? A Python script could probably be used to detect the presence of BuildKit. In a worst case scenario it tries to build an image with a `TARGETARCH`, tests to see if it was set, then proceeds or throws an error. I think that's a bloated but platform independent way to do it. I'm not sure how that fits in with your CI pipelines.

MatthewTingum, 2022-10-23:
I'm really only familiar with GitLab CI but I could see this being another job with an optional argument specifying architecture that overrides `TARGETARCH`.

rbs-jacob, 2022-10-24:
> That's an option, but then won't AARCH64 users find themselves with this same issue? Seems like it's just shifting the problem to different users.

This is true, and is what I meant by calling my proposed solution "brittle." My thinking was that this would at least get us back to where we were before M1 support was added, but your points are well-taken. 

The bit I mentioned about caching was just an aside – not actually too relevant to this discussion. Sorry for the confusion on that. 

Short-term, I think documenting this better is a good way to go. If you want to make a PR to improve the docs, I'd be happy to review it! Otherwise we can add it to the backlog. As you pointed out, at least one of the relevant places to edit is here:

https://github.com/redballoonsecurity/ofrak/blob/c0144b759cb2f06467ade1ede3c1f58194ecd763/docs/environment-setup.md?plain=1#L13

MatthewTingum, 2022-10-27:
Gotcha. I was trying to probe a little bit to see if there's a solution that might fit well with existing CI. I suppose that's over-scoping. I'll try to review the entirety of the docs over the next couple of days and put in a PR if this is the direction we're going.

What are your thoughts about this approach:
- Default `TARGETARCH` to `amd64` if not specified, as you mentioned
  - This covers (IMO) the broader spectrum of users as of writing
- Call out the need for BuildKit in the `environment-setup` doc in order to build for other supported architectures

Another approach:
- Immediately check if BuildKit is present by the presence of `TARGETARCH` or some other metric
  - If not, fail with a clear error

I'd like to say adding a required architecture option to `build_image.py` is the most verbose, impossible to miss solution. I need to read through the project to find out if that is completely true. I'd like to hear your opinion on this approach. This approach and the previous approach are not really universal and depend on one script being the only consumer / parent-of-consumer of `TARGETARCH`. If not, then maybe a master script comes into play, `ofrak build_image --config ofrak-ghidra.yml --base --finish `. Maybe I'm over-complicating this...

rbs-jacob, 2022-10-27:
Thanks for following up! I lean towards the first approach you mentioned that both makes a short-term fix for the most common use case and also improves the docs:

> * Default `TARGETARCH` to `amd64` if not specified, as you mentioned
>   * This covers (IMO) the broader spectrum of users as of writing
> * Call out the need for BuildKit in the `environment-setup` doc in order to build for other supported architectures

I'm fairly certain that the `TARGETARCH` argument/environment variable is used in exactly two places in the entire repo at this time – just two `Dockerstub` files used by the `build_image.py` script. As such, I'm thinking that enforcing an architecture in `build_image.py` might be overkill? In general, OFRAK isn't tied to a particular OS or architecture, beyond assuming the binaries of some dependencies are present at the expected paths (in particular [`toolchain.conf`](https://github.com/redballoonsecurity/ofrak/blob/master/ofrak_patch_maker/toolchain.conf). 

Perhaps @whyitfor or @EdwardLarson wants to weigh in?

MatthewTingum, 2022-11-23:
I got tied up in other things. I'll try to get back to this but consider me unassigned for the time being. Hopefully this issue serves as a troubleshooting resource for others until its fixed.

whyitfor, 2023-04-17:
This has come up again recently.

It would be good to make Docs very clear that DOCKER_BUILDKIT needs to be enabled, and check that all example instructions/Makefiles also use DOCKER_BUILDKIT.