redballoonsecurity/ofrak #156: LLVM_MACH_O_Parser

whyitfor, 2023-01-03:
**What is the use case for the feature?**
The `LLVM_MACH_O_Parser` would give the ability to parse Mach-O files.

**Does the feature contain any proprietary information about another company's intellectual property?**
No.

**How would you implement this feature?**
Continue working off of WIP code that was removed in #155.

**Are you interested in implementing it yourself?**
No. This is a good first contributor issue!

rbs-jacob, 2025-10-22:
Hi @saniyafatima07, thanks for your interest in contributing to OFRAK!

We'd love to add Mach-O support to OFRAK, but it might be a little less straightforward than this issue implies. The `LLVM_MACH_O_Parser` referred to here is just for the Patch Maker components (for advanced C patching), but won't actually enable OFRAK to unpack Mach-O files in general. Doing so will require extra changes.

To add Mach-O support, we first want to pull in one of these two libraries:

- [strongarm's Mach-O parser](https://github.com/datatheorem/strongarm)
- or [LIEF's Mach-O parser](https://lief.re/doc/latest/formats/macho/index.html)

Then we need to make unpacker and repacker components for Mach-O files. For example:

- https://github.com/redballoonsecurity/ofrak/blob/aaf2dab984c7719d181c3a841c98f7aa49768f33/ofrak_core/src/ofrak/core/elf/unpacker.py#L45-L58
- https://github.com/redballoonsecurity/ofrak/blob/aaf2dab984c7719d181c3a841c98f7aa49768f33/ofrak_core/src/ofrak/core/pe/unpacker.py#L21-L31

After that, we can add patch maker support for Mach-O files (via the `LLVM_MACH_O_Parser` mentioned in this issue). But that should be a separate pull request.

To summarize, if you're still interested in taking this on, you should start by adding a Mach-O unpacker using one of the libraries I linked above. Then open a pull request for review, and assign me. Please make sure that your pull request [follows our contributor guidelines](https://ofrak.com/docs/contributor-guide/getting-started.html). Thanks, and good luck!

saniyafatima07, 2025-10-23:
> Hi [@saniyafatima07](https://github.com/saniyafatima07), thanks for your interest in contributing to OFRAK!
> 
> We'd love to add Mach-O support to OFRAK, but it might be a little less straightforward than this issue implies. The `LLVM_MACH_O_Parser` referred to here is just for the Patch Maker components (for advanced C patching), but won't actually enable OFRAK to unpack Mach-O files in general. Doing so will require extra changes.
> 
> To add Mach-O support, we first want to pull in one of these two libraries:
> 
>     * [strongarm's Mach-O parser](https://github.com/datatheorem/strongarm)
> 
>     * or [LIEF's Mach-O parser](https://lief.re/doc/latest/formats/macho/index.html)
> 
> 
> Then we need to make unpacker and repacker components for Mach-O files. For example:
> 
>     * [ofrak/ofrak_core/src/ofrak/core/elf/unpacker.py](https://github.com/redballoonsecurity/ofrak/blob/aaf2dab984c7719d181c3a841c98f7aa49768f33/ofrak_core/src/ofrak/core/elf/unpacker.py#L45-L58)
>           
>           
>               Lines 45 to 58
>             in
>             [aaf2dab](/redballoonsecurity/ofrak/commit/aaf2dab984c7719d181c3a841c98f7aa49768f33)
>           
>         
>         
>           
>       
>               
>                 
>                  class ElfUnpacker(Unpacker[None]): 
>               
>       
>               
>                 
>                      id = b"ElfUnpacker" 
>               
>       
>               
>                 
>                      targets = (Elf,) 
>               
>       
>               
>                 
>                      children = ( 
>               
>       
>               
>                 
>                          ElfBasicHeader, 
>               
>       
>               
>                 
>                          ElfHeader, 
>               
>       
>               
>                 
>                          ElfProgramHeader, 
>               
>       
>               
>                 
>                          ElfSegment, 
>               
>       
>               
>                 
>                          ElfSectionHeader, 
>               
>       
>               
>                 
>                          ElfSection, 
>               
>       
>               
>                 
>                          ElfStringSection, 
>               
>       
>               
>                 
>                          ElfSymbolSection, 
>               
>       
>               
>                 
>                          CodeRegion, 
>               
>       
>               
>                 
>                      )
> 
>     * [ofrak/ofrak_core/src/ofrak/core/pe/unpacker.py](https://github.com/redballoonsecurity/ofrak/blob/aaf2dab984c7719d181c3a841c98f7aa49768f33/ofrak_core/src/ofrak/core/pe/unpacker.py#L21-L31)
>           
>           
>               Lines 21 to 31
>             in
>             [aaf2dab](/redballoonsecurity/ofrak/commit/aaf2dab984c7719d181c3a841c98f7aa49768f33)
>           
>         
>         
>           
>       
>               
>                 
>                  class PeUnpacker(Unpacker[None]): 
>               
>       
>               
>                 
>                      id = b"PeUnpacker" 
>               
>       
>               
>                 
>                      targets = (Pe,) 
>               
>       
>               
>                 
>                      children = ( 
>               
>       
>               
>                 
>                          PeMsDosHeader, 
>               
>       
>               
>                 
>                          PeFileHeader, 
>               
>       
>               
>                 
>                          PeOptionalHeader, 
>               
>       
>               
>                 
>                          PeDataDirectory, 
>               
>       
>               
>                 
>                          PeSectionHeader, 
>               
>       
>               
>                 
>                          PeSection, 
>               
>       
>               
>                 
>                      )
> 
> 
> After that, we can add patch maker support for Mach-O files (via the `LLVM_MACH_O_Parser` mentioned in this issue). But that should be a separate pull request.
> 
> To summarize, if you're still interested in taking this on, you should start by adding a Mach-O unpacker using one of the libraries I linked above. Then open a pull request for review, and assign me. Please make sure that your pull request [follows our contributor guidelines](https://ofrak.com/docs/contributor-guide/getting-started.html). Thanks, and good luck!

@rbs-jacob  Thank you for the reply. I will read about it and then give u a confirmation if I will be able to take it up.