redballoonsecurity/ofrak #186: Store known symbolic information in LinkableSymbols

marczalik, 2023-01-24:
**What is the use case for the feature?**
LinkableSymbols represent symbols in a program that OFRAK can link against with PatchMaker (i.e. a C patch you inject can call functions and access data defined by those functions). Previously these had to be always defined manually, even when some symbolic information was present that OFRAK could use. Adding symbolic information to LinkableSymbols will allow patch authors to more quickly link against known symbols. Furthermore, generating assembly stub files for all architectures for the Linker to link against will harmonize the method used by PatchMaker for linking.

**Does the feature contain any proprietary information about another company's intellectual property?**
No.

**How would you implement this feature?**
Symbolic information (vaddr, name, type, and mode) about `LinkableSymbols` for functions is currently stored in `ComplexBlock` objects. These `ComplexBlocks` can be tagged as `LinkableSymbols` and the information will be copied to these new resources via the `get_symbols()` method of `LinkableBinary`. `PatchFromSourceModifier` gets linkage info from `LinkableBinary`'s `make_linkable_bom()` method. `PatchFromSourceModifier` will be modified to generate assembly stub files for the Linker to link against using the linkage information taken from the `LinkableBinary`.

We will need to stress test these changes to ensure that performance bottlenecks are not introduced by providing the Linker with potentially hundreds or thousands of object files to link against. We should consider adding test cases to the CI pipeline.

**Are there any (reasonable) alternative approaches?**
This change will store `LinkableSymbol`s as descendant resources of a `LinkableBinary`. We should consider whether this is the best structure for this information and whether there are better alternatives.

Symbolic information will only be taken from the disassembler backend, not directly from the symtab. Disassemblers may disagree with the symtab (if present) when a binary has been compiled with position-independent code. However, this should not be an issue to use the linkage info from the disassembler as long as patch authors are compiling their patches to be position-independent.

**Are you interested in implementing it yourself?**
Yes.

**Potential post-MVP features**
- Creating LinkableSymbols for data. Currently it is not as easy to get symbolic information for data as it is for functions. Furthermore, we would need to handle each disassembler backend separately.

EdwardLarson, 2023-01-25:
> Adding symbolic information to LinkableSymbols will allow patch authors to more quickly link against known symbols.

The `LinkableSymbol` data structure already includes the the symbolic information necessary for the linker, what we need to do is automatically create/populate these data structures.

> Furthermore, generating assembly stub files for all architectures for the Linker to link against will harmonize the method used by PatchMaker for linking.

I think we can leave this out of the ticket, generating the stubs is out of scope for this ticket (they are already generated).

> PatchFromSourceModifier will be modified to generate assembly stub files for the Linker to link against using the linkage information taken from the LinkableBinary.

Doesn't need to be modified, it already does this. I think we can also add that the `LinkableBinary.get_symbols` method is how existing `LinkableSymbols` are passed to the stub-generating code

marczalik, 2023-01-26:
> Doesn't need to be modified, it already does this. I think we can also add that the LinkableBinary.get_symbols method is how existing LinkableSymbols are passed to the stub-generating code

Question about this: do the stubs get generated in `LinkableBinary.make_linkable_bom()` where stub files are being written out? If that's the case, what is the purpose of `generate_arm_stubs()` in `utils.py`? I don't see anywhere where this function is called.

marczalik, 2023-02-06:
To improve the efficiency of automatically extracting, creating stub files, assembling, and linking against symbols for large binaries with potentially thousands of functions, we will limit this process to those symbols that are required by the patch source to be linked against.

marczalik, 2023-03-16:
@EdwardLarson can this issue be closed now?

closed 2023-03-20 by whyitfor