redballoonsecurity/ofrak #204: PowerPC PatchMaker Support

EdwardLarson, 2023-02-02:
**What is the use case for the feature?**
Patching PPC binaries with PatchMaker requires support for PPC builds. The LLVM toolchain may be sufficient, but we should also have a GCC toolchain targeting PPC. For example, I'm not sure how good LLVM's support for the [VLE](https://www.nxp.com/docs/en/supporting-information/VLEPIM.pdf) encoding is.

**Does the feature contain any proprietary information about another company's intellectual property?**
Nope.

**How would you implement this feature?**
A Toolchain & tests that targets PPC. The same test could also test the PPC target for LLVM.

**Are there any (reasonable) alternative approaches?**
Just relying on LLVM is probably fine, but GCC toolchains give more flexibility.

**Are you interested in implementing it yourself?**
Not as the main contributor, but I'll definitely help out with blockers!

thejoelpatrol, 2023-02-20:
FWIW VLE might only be supported by NXP's fork of gcc

rbs-jacob, 2023-02-22:
> FWIW VLE might only be supported by NXP's fork of gcc

@thejoelpatrol This was a very helpful hint! Thanks for saving me a bunch of time. 

For any future folks discovering this issue: it looks like the NXP fork of GCC can be downloaded [here](https://www.nxp.com/design/software/development-software/s32-design-studio-ide/s32-design-studio-for-power-architecture:S32DS-PA), but that a sign-in is required. This makes it a bit complicated to have the [OFRAK PatchMaker `Dockerstub`](https://github.com/redballoonsecurity/ofrak/blob/master/ofrak_patch_maker/Dockerstub) automatically pull down the right version of the toolchain.

closed 2023-02-22 by andresito00

rbs-jacob, 2023-02-23:
My branch with VLE support can be found here: <https://github.com/redballoonsecurity/ofrak/compare/master...rbs-jacob:ofrak:feature/ppc-vle-toolchain>

It uses the NXP fork of GCC. I'm not making a PR anytime soon because of the sign-in mentioned above, and because I'm not sure what the rules are as far as distributing their fork. (I'd think because of GPL that I should be able to?)

commit c0b0cfd9 (2023-02-22): Add PPC GNU 10 Linux toolchain (fix #204) (#234)

* Add initial PPC GNU 10 Linux toolchain

* Pass monkey patch test

* Pass more tests

* Pass final remaining tests