redballoonsecurity/ofrak #241: `make inspect` fails on 6 tutorial `.ipynb` files.

ANogin, 2023-03-01:
**Which files would be affected?**
`ofrak_tutorial/notebooks_with_outputs/1_simple_string_modification.ipynb`
`ofrak_tutorial/notebooks_with_outputs/2_ofrak_internals.ipynb`
`ofrak_tutorial/notebooks_with_outputs/3_binary_format_modification.ipynb`
`ofrak_tutorial/notebooks_with_outputs/4_simple_code_modification.ipynb`
`ofrak_tutorial/notebooks_with_outputs/5_filesystem_modification.ipynb`
`ofrak_tutorial/notebooks_with_outputs/6_code_insertion_with_extension.ipynb`

**Does the proposed maintenance include non-doc string _functional_ changes to the Python code?**
Only in tutorial.

**Are you interested in implementing it yourself?**
Sure, trivial to change, but not sure:
- Whether we want this changed.
- Whether we want to do anything to reconcile the difference between what `make inspect` checks (when its jupyter support is installed?) vs what `pre-commit` and CI check.

whyitfor, 2023-03-15:
@ANogin where are you seeing these errors? What errors are you seeing?

We have a GitHub action that should be running `make inspect`, it appears to be working:  https://github.com/redballoonsecurity/ofrak/actions/runs/4431570151/jobs/7774675150

rbs-jacob, 2023-03-16:
Perhaps the discrepancy comes from different versions of `black` and/or `mypy`?

@ANogin if you do `python3 -m pip freeze`, do the versions match the [`requirements-test.txt`](https://github.com/redballoonsecurity/ofrak/blob/master/ofrak_core/requirements-test.txt) file?

ANogin, 2023-03-16:
It only checks them (and then fails) when `black[jupyter]` dependencies are installed.
```bash
$ docker run -it --rm python:3.8 /bin/bash
root@a22ddc70b7d5:/# git clone https://github.com/redballoonsecurity/ofrak.git; cd ofrak
[...]
root@a22ddc70b7d5:/ofrak# pip3 install -r ofrak_core/requirements-test.txt -r ofrak_core/requirements.txt
[...]
black . --check --diff
Skipping .ipynb files as Jupyter dependencies are not installed.
You can fix this by running ``pip install black[jupyter]``
All done! ✨ 🍰 ✨
406 files would be left unchanged.
root@a22ddc70b7d5:/ofrak# pip3 install -r ofrak_core/requirements-test.txt -r ofrak_core/requirements.txt 'black[jupyter]'
[...]
root@a22ddc70b7d5:/ofrak# make inspect
[...]
black . --check --diff
--- ofrak_tutorial/notebooks_with_outputs/1_simple_string_modification.ipynb    2023-03-16 23:35:37.852412 +0000:cell_4
+++ ofrak_tutorial/notebooks_with_outputs/1_simple_string_modification.ipynb    2023-03-16 23:39:33.111801 +0000:cell_4
@@ -1,6 +1,7 @@
 from ofrak.core import BinaryPatchModifier, BinaryPatchConfig
+
 
 async def replace_hello_world(ofrak_context, input_filename, output_filename):
     # Create the root resource from the input file
     root_resource = await ofrak_context.create_root_resource_from_file(input_filename)
 
@@ -10,12 +11,9 @@
     # Find the "Hello, World!" string in the data
     hello_world_offset = data.find(b"Hello, World!")
 
     # Patch the binary by writing the null-terminated string "Meow!" over
     # the first occurrence of "Hello, World!" using the BinaryPatchModifier
-    await root_resource.run(
-        BinaryPatchModifier,
-        BinaryPatchConfig(hello_world_offset, b"Meow!\0")
-    )
+    await root_resource.run(BinaryPatchModifier, BinaryPatchConfig(hello_world_offset, b"Meow!\0"))
 
     # Flush the modified program to disk
     await root_resource.flush_to_disk(output_filename)
would reformat ofrak_tutorial/notebooks_with_outputs/1_simple_string_modification.ipynb
[...]
would reformat ofrak_tutorial/notebooks_with_outputs/6_code_insertion_with_extension.ipynb

Oh no! 💥 💔 💥
6 files would be reformatted, 408 files would be left unchanged.
make: *** [Makefile:3: check-black] Error 1
```

whyitfor, 2023-03-20:
@ANogin, would you like to fix this?

It seems like two changes:
1. Updating `ofrak_tutorial` to install `black[jupyter]` with `ofrak_tutorial[test]`
2. Fixing the black issues in `ofrak_tutorial/notebooks_with_outputs/1_simple_string_modification.ipynb`

You can also update the precommit config file to specify [black-jupyter](https://black.readthedocs.io/en/stable/integrations/source_version_control.html).

ANogin, 2023-03-21:
Sure

closed 2023-03-23 by whyitfor