redballoonsecurity/ofrak #272: Building from scratch

morpheuslord, 2023-03-30:
I am trying to build ofrak from the source and I am facing the following issue.

[image]

How can I rectify this issue.

morpheuslord, 2023-03-30:
i just ran 

`pip install -e .`

rbs-jacob, 2023-03-30:
@EdwardLarson I was able to replicate this error by running `pip install .` in the root of the clone OFRAK source directory. 

@morpheuslord the recommended way to install OFRAK from source is [this section on installing from source in the OFRAK docs](https://ofrak.com/docs/environment-setup.html#from-source-code). That page goes into detail and contains all of the relevant information. I've included a summary of that information (tested on a fresh Debian system) below.

The gist is that you have to install `python3`, 'libmagic-dev', and `pip`—and `nodejs` and `npm` if you want to build the GUI—and then run `make develop` in the relevant directories. 

The install step on a Debian or Ubuntu system looks like the following.

``` bash
sudo apt install git git-lfs python3 python3-pip libmagic-dev nodejs npm
python3 -m pip install --upgrade pip setuptools wheel
git clone https://github.com/redballoonsecurity/ofrak.git
cd ofrak
```

The step to run the `make develop` in all of the OFRAK directories looks like the following. This first installs NPM dependencies, then finds all files named `Makefile` and runs `make develop` in their containing directory.

``` bash
for DIR in ofrak_type ofrak_io ofrak_patch_maker ofrak_core disassemblers/ofrak_angr disassemblers/ofrak_capstone; do make -C "${DIR}" develop; done
cd frontend && npm install && make develop && cd ..
```

morpheuslord, 2023-03-30:
i am doing this on windows is there any specific steps for windows

rbs-jacob, 2023-03-30:
@morpheuslord are you running commands from PowerShell, `cmd.exe`, or from bash via Cygwin or Windows Subsystem for Linux? Are you using a package manager like Choco or Winget, or no package manager?

We don't officially have OFRAK Windows support yet, although I have gotten it mostly working. There will be some upcoming GitHub issues, fixes, and other documentation related to Windows to be released shortly. In the meantime, there will be a few fixes [on this OFRAK on Windows pull request](https://github.com/redballoonsecurity/ofrak/pull/261).

morpheuslord, 2023-04-01:
ok

rbs-jacob, 2023-04-04:
@morpheuslord I responded to your other comment [here](https://github.com/redballoonsecurity/ofrak/pull/261#issuecomment-1496580254), but will copy the advice in this thread as well: run the following to fix the error you're having.

```
python -m pip uninstall python-magic python-magic-bin
python -m pip install --upgrade python-magic-bin
```

closed 2024-12-04 by whyitfor