redballoonsecurity/ofrak #311: Minor: `/tmp/ofrak.log` probably shouldn't be hardcoded

ANogin, 2023-05-19:
**What is the problem? (Here is where you provide a complete Traceback.)**

We probably should not assume we can have exclusive use of `/tmp/ofrak.log`. It may be somewhat safe in a controlled docker environment where there is likely at most one instance of OFRAK running at a time, but not with `pip install`, or whatever weird things people might choose to do, resulting in:

```
    ofrak = OFRAK()
  File "/ofrak_core/ofrak/ofrak_context.py", line 125, in __init__
    logging.getLogger().addHandler(logging.FileHandler(DEFAULT_OFRAK_LOG_FILE))
  File "/usr/local/lib/python3.7/logging/__init__.py", line 1087, in __init__
    StreamHandler.__init__(self, self._open())
  File "/usr/local/lib/python3.7/logging/__init__.py", line 1116, in _open
    return open(self.baseFilename, self.mode, encoding=self.encoding)
PermissionError: [Errno 13] Permission denied: '/tmp/ofrak.log'
```

**Please provide some information about your environment.**
At minimum we would like the following information on your platform and Python environment:
  - Platform: `redballoonsecurity/ofrak/ghidra`
  - Latest `master` branch.

**If you've discovered it, what is the root cause of the problem?**
`DEFAULT_OFRAK_LOG_FILE = os.path.join(tempfile.gettempdir(), "ofrak.log")` in `ofrak_core/ofrak/ofrak_context.py` seems to be hardcoded (as far as I can tell, the `DEFAULT_` part is misleading).

**How often does the issue happen?**
- Whenever '/tmp/ofrak.log' is not writable

**What are the steps to reproduce the issue?**
- Create '/tmp/ofrak.log' as root (e.g run ofrak as root), then run ofrak as non-root.
 
**How would you implement this fix?**
- Maybe at least include a process id in the log file name, or some such? Open to suggestions.

**Are there any (reasonable) alternative approaches?**
- Plenty

**Are you interested in implementing it yourself?**
- Sure - just let me know what behavior might make sense

whyitfor, 2023-05-20:
Interesting. I've never run into this issue. What steps did you take to create this situation where you did not have permission for `/tmp/ofrak.log`?

ANogin, 2023-05-20:
> Interesting. I've never run into this issue. What steps did you take to create this situation where you did not have permission for `/tmp/ofrak.log`?

Invoked the default docker setup of having ofrak run the ghidra server as root, then tried `docker exec -u aleksey` as I did not want the wrong ownership on the files mounted from host, and ran ofrak there.

But I would imagine multiple users running `pip install`ed ofrak on a shared server would result in the same issue.

whyitfor, 2023-05-26:
If seems like we might want to:

- have a default parameter, so that users can easily change it
- would having a convention such as `{whatever tem[}/ofrak_{timestamp}.log make sense for the default? I could see that being useful.

closed 2025-09-30 by whyitfor