redballoonsecurity/ofrak #317: FreespaceAnalyzer incorrectly caches free region information

rbs-alexr, 2023-05-25:
**What is the problem? (Here is where you provide a complete Traceback.)**
When I run the FreespaceAnalyzer on an Elf first (by viewing it as Allocatable), it correctly reports no free space. If I then create free space and then run the FreespaceAnalyzer again, it still reports having no free space. This is because the results are cached and have not been updated

**Please provide some information about your environment.**
At minimum we would like the following information on your platform and Python environment:
  - Platform:
    - If running in one of the OFRAK Docker images, say which image it is (for example, `redballoonsecurity/ofrak/ghidra`).
    - If not running in an OFRAK Docker image, please provide the output of `python3 -m platform`.
  - Python environment: Please provide the output of `python3 -m pip freeze`.
Running in the rbs-superdev image

<details open>
<summary>python3 -m pip freeze</summary>
<br>
  
  
```python
ailment==9.2.6
aiofiles==23.1.0
aiohttp==3.8.4
aiohttp-cors==0.7.0
aiosignal==1.3.1
alabaster==0.7.13
angr==9.2.6
archinfo==9.2.6
argon2-cffi==21.3.0
argon2-cffi-bindings==21.2.0
astunparse==1.6.3
async-generator==1.10
async-timeout==4.0.2
asynctest==0.13.0
attrs==23.1.0
autoflake==1.4
Babel==2.12.1
backcall==0.2.0
backoff==1.11.0
beartype==0.12.0
beautifulsoup4==4.12.2
binwalk==2.3.3+cddfede
bitstring==4.0.2
black==22.6.0
bleach==6.0.0
cached-property==1.5.2
cachetools==5.3.0
capstone==4.0.2
certifi==2023.5.7
cffi==1.15.1
charset-normalizer==2.0.12
claripy==9.2.6
cle==9.2.6
click==8.1.3
coverage==6.5.0
CppHeaderParser==2.7.4
cryptography==3.4.8
dataclasses==0.6
debugpy==1.6.7
decorator==5.1.1
defusedxml==0.7.1
dill==0.3.5.1
docutils==0.18.1
dpkt==1.9.8
entrypoints==0.4
exceptiongroup==1.1.1
execnet==1.9.0
fastjsonschema==2.16.3
fdt==0.3.3
frozenlist==1.3.3
fun-coverage==0.2.0
future==0.18.3
ghp-import==2.1.0
[... 211 lines cut ...]
    assert len(allocatable.free_space_ranges) == 0
    data_length = await resource_under_test.get_data_length()
    config = PartialFreeSpaceModifierConfig(
        MemoryPermissions.RX,
        range_to_remove=Range.from_size(0, data_length - 10),
        fill=b"\xfe\xed\xfa\xce",
    )
    await resource_under_test.run(PartialFreeSpaceModifier, config)
    free_space = await resource_under_test.get_only_child_as_view(FreeSpace)
    free_space_data = await free_space.resource.get_data()
    assert free_space_data == config.fill + (b"\x00" * (data_length - 10 - len(config.fill)))
    allocatable = await resource_under_test.view_as(Allocatable)
    assert allocatable.free_space_ranges[MemoryPermissions.RX] == [Range(0x0, 0xf6)]
```

It fails with a KeyError for the last assert because free_space_ranges is {}

**How would you implement this fix?**
Honestly not totally sure, but the FreeSpaceAnalyzer should be re-run every time the resource is viewed as Allocatable rather than use the cached values. Maybe this involves removing the Allocatable tag from the resource at the end of the FreeSpaceAnalyzer so that the Analyzer is re-run.

**Are there any (reasonable) alternative approaches?**
Not that I am aware of

**Are you interested in implementing it yourself?**
I don't think I would be very efficient/helpful fixing this given my limited knowledge of the internals of OFRAK.

whyitfor, 2023-05-26:
One possible stopgap mitigation would be to have free space modifiers invalidate Allocatables if they exist. This doesn't seem to work. I tried adding this to the end of the `PartialFreeSpaceModifier`:
```
        child = await mem_region_view.resource.create_child_from_view(
            FreeSpace(
                freed_range.start,
                freed_range.length(),
                config.permissions,
            ),
            data_range=patch_range,
        )
        try:
            allocatable = await child.get_only_ancestor_as_view(
                Allocatable, r_filter=ResourceFilter.with_tags(Allocatable)
            )
            allocatable.resource.remove_attributes(AttributesType[Allocatable])
        except NotFoundError:
            pass
```

It results in an error:
```
    @classmethod
    def create(cls: Type[RV], resource_model: ResourceModel) -> RV:
        attributes_fields_values: Dict[str, Any] = dict()
        for required_attrs_type in cls.composed_attributes_types:
            if required_attrs_type not in resource_model.attributes:
>               raise ValueError(
                    f"Required attributes type {required_attrs_type.__name__} not "
                    f"provided in {resource_model.attributes}"
                )
E               ValueError: Required attributes type AttributesType[Allocatable] not provided in {<class 'ofrak.model._auto_attributes.AttributesType[Addressable]'>: AttributesType[Addressable](virtual_address=0), <class 'ofrak.model._auto_attributes.AttributesType[MemoryRegion]'>: AttributesType[MemoryRegion](size=256), <class 'ofrak.model.resource_model.Data'>: Data(_offset=0, _length=256)}
```

whyitfor, 2023-05-26:
Appending the following to the end of the `PartialFreeSpaceModifier` does seem to fix the test:

```
        try:
            allocatable = await child.get_only_ancestor_as_view(
                Allocatable, r_filter=ResourceFilter.with_tags(Allocatable)
            )
            perm_list = allocatable.free_space_ranges.get(config.permissions)
            if perm_list is None:
                perm_list = list()
                allocatable.free_space_ranges[config.permissions] = perm_list
            perm_list.append(freed_range)
        except NotFoundError:
            pass
```

This might be a way forward. Should `Allocatable` have an update method to add new ranges to it (and perform this saving)?