redballoonsecurity/ofrak #353: Create root resource from directory

csisl, 2023-07-28:
**What is the use case for the feature?**

If a file has already been extracted, let the user create a resource for the root directory of that output.  

An example scenario:  

A compressed file is extracted, but not recursively. The process ends and only some files have been extracted. Later, if the user wants to extract other files, they would have to start over by unpacking the original compressed file.  

Rather than re-extracting from the top-level, let the user resume where they left off and navigate the resource by creating a resource from a root directory.  

**Does the feature contain any proprietary information about another company's intellectual property?**

No

**How would you implement this feature?**

Inside of `ofrak_context.py`, implement a `create_root_resource_from_directory('/path/to/dir')` method. Then the method `filesystem.py:initialize_from_disk('/path/to/dir')` can be called so that all children and grandchildren in the directory can be registered.

This would make it easy to resume extraction if the process spans multiple runs.

whyitfor, 2023-07-28:
I like this idea! I actually found myself wishing this was possible recently as well.

@EdwardLarson, @rbs-jacob, what are your thoughts on adding this higher-level API?

EdwardLarson, 2023-07-28:
Yup, this is sensible and straightforward, I like it.

whyitfor, 2023-07-28:
@csisl, do you have any interest in implementing this?

Figured giving you first dibs makes sense as it was your idea:)

csisl, 2023-07-29:
Thank you for asking, but you guys can go ahead