redballoonsecurity/ofrak #368: error while installing

shashankx86, 2023-08-30:
Building base.Dockerfile, finish.Dockerfile for ghidra from: ofrak_type, ofrak_io, ofrak_patch_maker, ofrak_core, disassemblers/ofrak_ghidra, frontend.
base.Dockerfile built.
finish.Dockerfile built.
[+] Building 2.9s (3/3) FINISHED                                                                                                                                                                                             docker:default
 => [internal] load build definition from finish.Dockerfile                                                                                                                                                                            0.1s
 => => transferring dockerfile: 1.06kB                                                                                                                                                                                                 0.0s
 => [internal] load .dockerignore                                                                                                                                                                                                      0.1s
 => => transferring context: 128B                                                                                                                                                                                                      0.0s
 => ERROR [internal] load metadata for docker.io/redballoonsecurity/ofrak/ghidra-base:e70519e5                                                                                                                                         2.7s
------
 > [internal] load metadata for docker.io/redballoonsecurity/ofrak/ghidra-base:e70519e5:
------
finish.Dockerfile:1
--------------------
   1 | >>> FROM redballoonsecurity/ofrak/ghidra-base:e70519e5
   2 |     
   3 |     ARG OFRAK_SRC_DIR=/
--------------------
ERROR: failed to solve: redballoonsecurity/ofrak/ghidra-base:e70519e5: pull access denied, repository does not exist or may require authorization: server message: insufficient_scope: authorization failed
Error running command: 'docker build -t redballoonsecurity/ofrak/ghidra:e70519e5 -t redballoonsecurity/ofrak/ghidra:latest -f finish.Dockerfile --build-arg INSTALL_TARGET=develop .'
Exit status: 1

EdwardLarson, 2023-08-30:
Hi there, thanks for posting! Can you paste the command you used to build the image, including all the arguments to `build_image.py`? If building from scratch you need to including both `--base` and `--finish`, from this error message my best guess is that the `--base` flag is missing.

The command to build the Ghidra-only image is `python3 build_image.py --config ofrak-ghidra.yml --base --finish`
Use [this](https://ofrak.com/docs/environment-setup.html#docker) as a reference and check in here if any of the steps there are not working.

shashankx86, 2023-08-31:
> `python3 build_image.py --config ofrak-ghidra.yml --base --finish`

already done once, got same error when ran `python3 build_image.py --config ofrak-ghidra.yml --base --finish`
docs said if once image is built then then there is no need use `--base`

EdwardLarson, 2023-08-31:
This looks like it could be related to the problems in this thread: https://github.com/docker/compose/issues/8449 (that has other Docker-compose-specific issues intersecting it, but it deals with the same `pull access denied, repository does not exist or may require authorization: server message: insufficient_scope: authorization failed` error). I would suggest trying to run with the Docker buildkit turned off (prefix the command with `DOCKER_BUILDKIT=0`), and although that should still eventually fail to build as some of the dependencies require buildkit features, it may get past the initial metadata loading and help narrow down the error.

OFRAK issue #83 has some more info on the buildkit problems and consequences.

Can you provide some info about your system? For example, is it an M1 Mac, what OS, etc. As many of the bug report template fields as possible please.

shashankx86, 2023-09-01:
> Can you provide some info about your system? For example, is it an M1 Mac, what OS, etc. As many of the bug report template fields as possible please.

i3-8gen, Debian 12.1, kernel: 6.1.0.11

shashankx86, 2023-09-23:
used conda, not using docker image anymore.

closed 2023-09-23 by shashankx86