redballoonsecurity/ofrak #380: File and Directory upload, simple text editor.

shashankx86, 2023-09-20:
## File and Directory upload

**What is the use case for the feature?**
after unpacking firmware there should option to add new files and Directory

**Does the feature contain any proprietary information about another company's intellectual property?**
IDK

**How would you implement this feature?**
there should be a button to add new file and Directory in extracted firmware 

**Are there any (reasonable) alternative approaches?**
IDK

**Are you interested in implementing it yourself?**
maybe, I guess 

## simple text editor

**What is the use case for the feature?**
There should be option to open file in simple text editor in GUI
like in my case I was working with android firmware and i had to modify buil.prop file, it contains readable text and can be edited with simple text editor

**Does the feature contain any proprietary information about another company's intellectual property?**
IDK

**How would you implement this feature?**
there should be a option in modify button in which way user wants to open that file

**Are there any (reasonable) alternative approaches?**
IDK

**Are you interested in implementing it yourself?**
maybe, I guess

dannyp303, 2023-09-20:
Hello! 

There are ways to do this in OFRAK, it is just not well supported in the GUI.
See [FilesystemRoot.addfile](https://github.com/redballoonsecurity/ofrak/blob/c7088dede76305642cc81b23fab7b4a20997e947/ofrak_core/ofrak/core/filesystem.py#L564C16-L564C16). 

HINT: To access FilesystemRoot methods you have to call `resource.view_as(FilesystemRoot)` on the appropriate resource first.  (Checking out the [tests](https://github.com/redballoonsecurity/ofrak/blob/c7088dede76305642cc81b23fab7b4a20997e947/ofrak_core/test_ofrak/components/test_general_filesystems.py#L83) could help.)

If you're looking for a quick route to doing what you want in the GUI I would suggest using the `Show Script` and `Run Script` features (more info can be found on our blog [here](https://redballoonsecurity.com/ofrak-310/) and [here](https://redballoonsecurity.com/ofrak-3-0-0/)). Modifying your auto-generated script is a great way to get started with the OFRAK python API (the way OFRAK was intended to be used).

If you have any ideas on how you would better implement this in the frontend feel free to submit a PR!

rbs-jacob, 2023-09-20:
I agree that it would be nice to have! 

In the meantime, you can download a file, change it offline, and re-upload to achieve the same thing. 

https://github.com/redballoonsecurity/ofrak/assets/99368685/11521e5b-e682-4ec6-87ba-ba3d0a03084b

shashankx86, 2023-09-21:
what i do about this?

Using the Python implementation of the Shannon entropy calculation! This is potentially very slow, and is only used when the C extension cannot be built/found.

dannyp303, 2023-09-21:
> untitled.mp4 
> for some reason my file (kinda) get merged @rbs-jacob @dannyp303

Thanks! Addressed in #381

shashankx86, 2023-09-24:
@rbs-jacob is there a way to extract firmware file to specific folder then modify firmware in from that folder and then repack it

closed 2023-10-04 by shashankx86