redballoonsecurity/ofrak #389: java is_tool_installed() not working correctly.

sprout42, 2023-10-18:
**What is the problem? (Here is where you provide a complete Traceback.)**
The dependency check for `java` using the `--help` argument does not result in a successful return code which prevents a properly installed `java` from being detected.
```
$ ofrak deps --missing-only
[ ] apktool
        https://ibotpeaches.github.io/Apktool/
        [ApkPacker, ApkUnpacker]
[ ] binwalk
        https://github.com/ReFirmLabs/binwalk
        [BinwalkAnalyzer]
[ ] java
        https://openjdk.org/projects/jdk/11/
        [ApkPacker]
[ ] python-lzo
        https://github.com/jd-boyd/python-lzo
        [UbiPacker, UbiIdentifier, UbifsUnpacker, UbiUnpacker, UbiAnalyzer, UbifsIdentifier, UbifsAnalyzer]

$ java --help
Unrecognized option: --help
Error: Could not create the Java Virtual Machine.
Error: A fatal exception has occurred. Program will exit.
```

**Please provide some information about your environment.**
- Ubuntu 22.04.03 LTS
```
$ python3 -m platform
Linux-5.15.0-83-generic-x86_64-with-glibc2.35

$ python3 -c "import sys;print(sys.executable)"
/usr/bin/python3
```
- `openjdk-11-jdk` installed through `apt`
- `ofrak` installed through `pip` in a clean virtual environment
```
python -m pip freeze
aiohttp==3.8.6
aiohttp-cors==0.7.0
aiosignal==1.3.1
async-timeout==4.0.3
attrs==23.1.0
beartype==0.12.0
black==23.3.0
cffi==1.16.0
charset-normalizer==3.3.0
click==8.1.7
cstruct==5.2
fdt==0.3.3
frozenlist==1.4.0
gitdb==4.0.10
GitPython==3.1.32
idna==3.4
immutabledict==2.2.0
importlib-metadata==6.8.0
intervaltree==3.1.0
jefferson==0.4.5
keystone-engine==0.9.2
lief==0.12.3
lzallright==0.2.3
multidict==6.0.4
mypy-extensions==1.0.0
ofrak==3.2.0.post0
ofrak-io==1.1.0
ofrak-patch-maker==4.0.2
ofrak-type==2.2.0
orjson==3.8.14
packaging==23.2
pathspec==0.11.2
pefile==2023.2.7
platformdirs==3.11.0
pycdlib==1.12.0
pycparser==2.21
python-magic==0.4.27
reedsolo==1.7.0
smmap==5.0.1
sortedcontainers==2.2.2
synthol==0.1.1
tomli==2.0.1
typeguard==2.13.3
typing-inspect==0.7.1
typing_extensions==4.8.0
ubi-reader==0.8.5
xattr==0.10.1
yarl==1.9.2
zipp==3.17.0
```

**If you've discovered it, what is the root cause of the problem?**

The `--help` option provided as the `install_check_arg` isn't supported by `java` resulting in a return code of 1 from the `java`
https://github.com/redballoonsecurity/ofrak/blob/6aa48f99a63a0a5f8fd37101f16d1b38172a1305/ofrak_core/ofrak/core/apk.py#L29C1-L29C1
```
$ java --help &> /dev/null
$ echo $?
1
```
If the tool is run with `-help` instead it will return a success value of 0
```
$ java -help &> /dev/null
$ echo $?
0
```

**How often does the issue happen?**
every time

**What are the steps to reproduce the issue?**
1. install `openjdk-11-jdk` through apt
2. install `ofrak`
3. run `ofrak deps --missing-only` 

**How would you implement this fix?**
change the `"--help"` `install_check_arg` to `-help`:
```
JAVA = ComponentExternalTool(
    "java",
    "https://openjdk.org/projects/jdk/11/",
    "-help",
    apt_package="openjdk-11-jdk",
    brew_package="openjdk@11",
)
```

**Are there any (reasonable) alternative approaches?**
This seems pretty simple and straight forward.

**Are you interested in implementing it yourself?**
Not at the moment.

rbs-jacob, 2023-10-18:
Hey @sprout42! Thanks for reporting this :)

Happy to do a quick fix for this. Before pushing a change, I'll try with a couple different `java` versions to try and pick something portable so that this doesn't pop up again in the future.

Worth noting that this has also manifested when trying to run `ofrak deps` on Windows.

rbs-jacob, 2023-10-18:
Interestingly, I wasn't able to replicate the failure on JDK 11, but was able to replicate it on JDK 8. Either way, the suggested fix is confirmed to be a good one! Will follow up with a PR shortly.

``` bash
for CMD in -h --help -help; do 
  echo "${CMD}"
  for JAVA in /usr/lib/jvm/*/bin/java; do 
    "${JAVA}" "${CMD}" >/dev/null 2>&1 \
      && printf '\t\033[1;32mWorked\033[1;m  ' \
      || printf '\t\033[1;31mFailed\033[1;m  '
    echo "${JAVA}"
  done
done
```

[image]

closed 2023-10-23 by rbs-jacob