redballoonsecurity/ofrak #430: crashes when checking for missing dependencies

hikir1, 2024-02-22:
Ofrak crashes when invoked with `-x` and apktool is missing. It's method for checking for the existence of apktool involves calling it, which fails and leads to the error:
> PermissionError: [Errno 13] Permission denied: 'apktool'

Solution: check if apktool is in PATH

To reproduce, run `ofrak gui -x` without apktool installed.

Stack trace:
```
  File "/home/hikir1/.local/bin/ofrak", line 33, in <module>
    sys.exit(load_entry_point('ofrak', 'console_scripts', 'ofrak')())
  File "/home/hikir1/ofrak/ofrak_core/ofrak/__main__.py", line 15, in main
    ofrak_cli.parse_and_run(sys.argv[1:])
  File "/home/hikir1/ofrak/ofrak_core/ofrak/cli/ofrak_cli.py", line 221, in parse_and_run
    parsed.run(parsed)
  File "/home/hikir1/ofrak/ofrak_core/ofrak/cli/ofrak_cli.py", line 182, in run
    ofrak.run(self.ofrak_func, args)
  File "/home/hikir1/ofrak/ofrak_core/ofrak/ofrak_context.py", line 197, in run
    asyncio.get_event_loop().run_until_complete(self.run_async(func, *args))
  File "/usr/lib/python3.9/asyncio/base_events.py", line 642, in run_until_complete
    return future.result()
  File "/home/hikir1/ofrak/ofrak_core/ofrak/ofrak_context.py", line 187, in run_async
    ofrak_context = await self.create_ofrak_context()
  File "/home/hikir1/ofrak/ofrak_core/ofrak/ofrak_context.py", line 163, in create_ofrak_context
    components = await self._get_discovered_components()
  File "/home/hikir1/ofrak/ofrak_core/ofrak/ofrak_context.py", line 221, in _get_discovered_components
    await asyncio.gather(
  File "/home/hikir1/ofrak/ofrak_core/ofrak/model/component_model.py", line 61, in is_tool_installed
    proc = await asyncio.create_subprocess_exec(
  File "/usr/lib/python3.9/asyncio/subprocess.py", line 236, in create_subprocess_exec
    transport, protocol = await loop.subprocess_exec(
  File "/usr/lib/python3.9/asyncio/base_events.py", line 1661, in subprocess_exec
    transport = await self._make_subprocess_transport(
  File "/usr/lib/python3.9/asyncio/unix_events.py", line 197, in _make_subprocess_transport
    transp = _UnixSubprocessTransport(self, protocol, args, shell,
  File "/usr/lib/python3.9/asyncio/base_subprocess.py", line 36, in __init__
    self._start(args=args, shell=shell, stdin=stdin, stdout=stdout,
  File "/usr/lib/python3.9/asyncio/unix_events.py", line 789, in _start
    self._proc = subprocess.Popen(
  File "/usr/lib/python3.9/subprocess.py", line 951, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/usr/lib/python3.9/subprocess.py", line 1823, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
```

- Platform: Linux-5.10.0-22-amd64-x86_64-with-glibc2.31
- Python environment:
  - ailment==9.2.6
  - aiohttp==3.8.6
  - aiohttp-cors==0.7.0
  - aiosignal==1.3.1
  - angr==9.2.6
  - archinfo==9.2.6
  - argon2-cffi==23.1.0
  - argon2-cffi-bindings==21.2.0
  - asciimatics==1.14.0
  - astroid==2.5.1
  - asttokens==2.4.1
  - async-timeout==4.0.3
  - attrs==20.3.0
  - autoflake==1.4
  - beartype==0.12.0
  - beautifulsoup4==4.9.3
  - binwalk==2.2.1
  - bitarray==2.9.2
  - bitstring==4.1.4
  - black==23.3.0
  - bleach==6.1.0
  - cached-property==1.5.2
  - cachetools==5.3.2
  - capstone==4.0.2
  - certifi==2020.6.20
  - cffi==1.16.0
  - chardet==4.0.0
  - charset-normalizer==3.3.2
  - claripy==9.2.6
  - cle==9.2.6
  - click==8.1.7
  - comm==0.2.1
  - coverage==6.5.0
  - CppHeaderParser==2.7.4
  - cryptography==3.3.2
  - cstruct==5.3
  - cycler==0.10.0
  - debugpy==1.8.1
  - decorator==5.1.1
  - defusedxml==0.7.1
  - distro==1.5.0
  - docker==4.1.0
  - docker-compose==1.25.0
  - dockerpty==0.4.1
  - docopt==0.6.2
  - dpkt==1.9.8
  - exceptiongroup==1.2.0
  - execnet==2.0.2
  - executing==2.0.1
  - fastjsonschema==2.19.1
  - fdt==0.3.3
  - frozenlist==1.4.1
  - fun-coverage==0.2.0
  - future==0.18.3
  - ghp-import==2.1.0
  - gitdb==4.0.11
  - GitPython==3.1.41
  - greenlet==3.0.3
  - html5lib==1.1
  - httplib2==0.18.1
  - hypothesis==6.39.6
  - hypothesis-trio==0.6.0
  - idna==2.10
  - immutabledict==2.2.0
  - importlib-metadata==4.13.0
  - importlib-resources==6.1.1
  - iniconfig==2.0.0
  - intervaltree==3.1.0
  - ipykernel==6.29.2
  - ipython==8.18.1
  - ipython-genutils==0.2.0
  - isort==5.6.4
  - itanium-demangler==1.1
  - jedi==0.19.1
  - jefferson==0.4.5
  - Jinja2==3.0.0
  - jsonschema==3.2.0
  - jupyter_client==8.6.0
  - jupyter_core==5.7.1
  - jupyterlab_pygments==0.3.0
  - keystone-engine==0.9.2
  - kiwisolver==1.3.1
  - lazy-object-proxy==0.0.0
  - lief==0.12.3
  - logilab-common==1.8.1
  - lxml==4.6.3
  - lzallright==0.2.4
  - Markdown==3.5.2
  - MarkupSafe==2.1.5
  - matplotlib==3.3.4
  - matplotlib-inline==0.1.6
  - mccabe==0.6.1
  - mergedeep==1.3.4
  - mistune==3.0.1
  - mkdocs==1.2.3
  - mkdocs-autorefs==0.3.0
  - mkdocs-gen-files==0.3.3
  - mkdocs-literate-nav==0.4.0
  - mkdocs-material==7.3.3
  - mkdocs-material-extensions==1.3.1
  - mkdocstrings==0.16.2
  - more-itertools==4.2.0
  - mpmath==1.3.0
  - mulpyplexer==0.9
  - multidict==6.0.5
  - mypy==0.942
  - mypy-extensions==0.4.3
  - nampa==0.1.1
  - nbclient==0.9.0
  - nbconvert==7.16.1
  - nbformat==5.9.2
  - nbval==0.9.6
  - nest-asyncio==1.6.0
  - networkx==3.2.1
  - notebook==6.4.13
  - numpy==1.19.5
  - -e git+ssh://git@github.com/redballoonsecurity/ofrak.git@f05d3ca180cb546c42116781f6cb8a3d943d9dc5#egg=ofrak&subdirectory=ofrak_core
  - -e git+ssh://git@github.com/redballoonsecurity/ofrak.git@f05d3ca180cb546c42116781f6cb8a3d943d9dc5#egg=ofrak_angr&subdirectory=disassemblers/ofrak_angr
  - -e git+ssh://git@github.com/redballoonsecurity/ofrak.git@f05d3ca180cb546c42116781f6cb8a3d943d9dc5#egg=ofrak_binary_ninja&subdirectory=disassemblers/ofrak_binary_ninja
  - -e git+ssh://git@github.com/redballoonsecurity/ofrak.git@f05d3ca180cb546c42116781f6cb8a3d943d9dc5#egg=ofrak_capstone&subdirectory=disassemblers/ofrak_capstone
  - -e git+ssh://git@github.com/redballoonsecurity/ofrak.git@f05d3ca180cb546c42116781f6cb8a3d943d9dc5#egg=ofrak_ghidra&subdirectory=disassemblers/ofrak_ghidra
  - -e git+ssh://git@github.com/redballoonsecurity/ofrak.git@f05d3ca180cb546c42116781f6cb8a3d943d9dc5#egg=ofrak_io&subdirectory=ofrak_io
  - -e git+ssh://git@github.com/redballoonsecurity/ofrak.git@f05d3ca180cb546c42116781f6cb8a3d943d9dc5#egg=ofrak_patch_maker&subdirectory=ofrak_patch_maker
  - -e git+ssh://git@github.com/redballoonsecurity/ofrak.git@f05d3ca180cb546c42116781f6cb8a3d943d9dc5#egg=ofrak_tutorial&subdirectory=ofrak_tutorial
  - -e git+ssh://git@github.com/redballoonsecurity/ofrak.git@f05d3ca180cb546c42116781f6cb8a3d943d9dc5#egg=ofrak_type&subdirectory=ofrak_type
  - olefile==0.46
  - orjson==3.8.14
  - outcome==1.3.0.post0
  - packaging==23.1
  - pandocfilters==1.5.1
  - parso==0.8.3
  - pathspec==0.11.2
  - pefile==2023.2.7
  - pexpect==4.9.0
  - Pillow==10.0.0
  - platformdirs==3.10.0
  - pluggy==1.4.0
  - plumbum==1.8.2
  - ply==3.11
  - present==0.6.0
  - progressbar2==4.3.2
  - prometheus_client==0.20.0
  - prompt-toolkit==3.0.43
  - protobuf==3.20.3
  - psutil==5.9.8
  - ptyprocess==0.7.0
  - pure-eval==0.2.2
  - py==1.11.0
  - pycdlib==1.12.0
  - pycparser==2.21
  - pycurl==7.43.0.6
  - pyelftools==0.29
  - pyfiglet==1.0.2
  - pyflakes==3.2.0
  - Pygments==2.17.2
  - PyGObject==3.38.0
  - pylint==2.7.2
  - pymdown-extensions==9.11
  - PyOpenGL==3.1.5
  - pyOpenSSL==20.0.1
  - pyparsing==2.4.7
  - PyQt5==5.15.2
  - PyQt5-sip==12.8.1
  - pyqtgraph==0.11.1
  - pyrsistent==0.15.5
  - PySimpleSOAP==1.16.2
  - PySMT==0.9.6.dev53
  - pytest==7.1.3
  - pytest-aiohttp==1.0.5
  - pytest-asyncio==0.19.0
  - pytest-cov==4.1.0
  - pytest-lazy-fixture==0.6.3
  - pytest-xdist==3.5.0
  - python-apt==2.2.1
  - python-dateutil==2.8.2
  - python-debian==0.1.39
  - python-debianbts==3.1.0
  - python-magic==0.4.27
  - python-utils==3.8.2
  - pytkdocs==0.12.0
  - pyvex==9.2.6
  - PyYAML==6.0.1
  - pyyaml_env_tag==0.1
  - pyzmq==25.1.2
  - reedsolo==1.7.0
  - reportbug==7.10.3+deb11u1
  - requests==2.25.1
  - rpyc==5.3.1
  - scipy==1.6.0
  - scour==0.38.2
  - Send2Trash==1.8.2
  - six==1.16.0
  - smmap==5.0.1
  - sniffio==1.3.0
  - sortedcontainers==2.2.2
  - soupsieve==2.2.1
  - stack-data==0.6.3
  - sympy==1.12
  - synthol==0.1.1
  - termcolor==1.1.0
  - terminado==0.18.0
  - texttable==1.6.3
  - tinycss2==1.2.1
  - tokenize-rt==5.2.0
  - toml==0.10.1
  - tomli==2.0.1
  - tornado==6.4
  - traitlets==5.14.1
  - trio==0.24.0
  - trio_asyncio==0.14.0
  - typed-ast==1.4.2
  - typeguard==2.13.3
  - typing-inspect==0.7.1
  - typing_extensions==4.8.0
  - ubi-reader==0.8.5
  - unicorn==1.0.2rc4
  - urllib3==1.26.5
  - watchdog==4.0.0
  - wcwidth==0.2.6
  - webencodings==0.5.1
  - websocket-client==0.57.0
  - wrapt==1.12.1
  - xattr==0.10.1
  - yarl==1.9.4
  - z3-solver==4.12.5.0
  - zipp==3.17.0

whyitfor, 2024-02-26:
The error message you post -- `PermissionError: [Errno 13] Permission denied: 'apktool'` -- seems to indicate a permission error when running `apktool`, and not a problem with the command not existing.

Is it possible that what was encountered here was a permission error?

hikir1, 2024-02-26:
It looks like their was a directory owned by another user on my PATH that I don't have access to. Perhaps ofrak could catch permission errors when searching for dependencies and output a warning?

whyitfor, 2024-04-19:
@hikir1, is this issue addressed with #432 and #433? Is there any remaining work?

hikir1, 2024-05-14:
The test isn't working for some reason and I haven't had much time to figure out why.