redballoonsecurity/ofrak #476: Fall back on `gzip` if `pigz` fails

rbs-jacob, 2024-06-14:
Recent PR #472 uses Python's built-in `gzip` module for decompressing GZIP data if possible, and falls back on the system version of `pigz` if the Python version fails. 

Inside of OFRAK Docker images, we can be confident that `pigz` is present. But when a user installs OFRAK via `pip install`, they may not have `pigz` on their system. They are, however, likely to have `gzip` on their system. As such, we should fall back from Python `gzip` to `pigz` to `gzip` to increase the likelihood that *something* will work for GZIP decompression. 

Implementing this may involve modifying `ComponentExternalTool` ([see this comment on #472](https://github.com/redballoonsecurity/ofrak/pull/472#issuecomment-2151308732)). This may have implications for other OFRAK functionality using `ComponentExternalTool`.

alchzh, 2024-07-16:
If we're only using pigz as a fallback for the trailing data case and prefer python standard library gzip otherwise, I think it makes more sense to implement a fix in python ourselves, it's a pretty simple modification to the standard library package. (see this stackoverflow answer: https://stackoverflow.com/questions/4928560/how-can-i-work-with-gzip-files-which-contain-extra-data)

there isn't any sort of official Windows packaging for `pigz`

whyitfor, 2024-07-16:
Like the idea of removing dependencies if we can. If your solution passes all the tests (including https://github.com/redballoonsecurity/ofrak/blob/master/ofrak_core/test_ofrak/components/test_gzip_component.py#L38, then it should work).

Would also make sense to update our testing code to use python std. library to set up/validate zip files for testing.

rbs-jacob, 2024-07-16:
One issue not taken into account in #472, when we switched the default GZIP unpacker to the Python standard library instead of `pigz`, is that `pigz` is almost certainly faster for large files where the constant cost of starting a process is negligible compared to the processing time. 

If, in fact, we can reliably use Python standard library `gzip` to correctly handle trailing garbage data as that StackOverflow post implies, then probably the logic should be something like:

- If file over size threshold and `pigz` is installed:
    - Inflate with `pigz`
- Else
    - Inflate with Python standard library `gzip`

alchzh, 2024-09-03:
Can be closed after #485 merged. Should have written "Fixes" in that PR 🤦

closed 2024-09-03 by rbs-jacob