redballoonsecurity/ofrak #534: Adding CHANGELOG.md entries in pull requests is not fun

alchzh, 2024-12-13:
Currently, every pull request causes a merge conflict on `CHANGELOG.md` files if a changelog entry was added to the same section since the pull request branch was forked off. This is a well known annoyance with git and requires a manual merge conflict resolution right before a pull request is merged. Coordination is required here since no other pull request can be merged in the meantime without causing another conflict, and the whole test suite is triggered just from the one line addition so you have to wait a while after the push to merge it.

There are a few steps we can consider to smooth this whole 
- Adopting a tool like [`towncrier`](https://pypi.org/project/towncrier/) to manage the changelog. This works better for projects with frequent release cycles though, which we currently aren't doing for OFRAK.
- Adding a git custom merge strategy. This wouldn't work with the github UI but would make the merge automatic instead of a conflict resolution
- Adding `CHANGELOG.md` files to `path_ignore` in Github actions. This should at least prevent us from having to wait for the whole test suite to run after a changelog update before merging in a branch.

**Which files would be affected?**
Possible files affected:
- git configuration files
- `CHANGELOG.md` files
- github workflow config

**Does the proposed maintenance include non-doc string _functional_ changes to the Python code?**
No

**Are you interested in implementing it yourself?**
Yes

whyitfor, 2025-01-02:
@alchzh, seems like the `path_ignore` option is a good first step. Can we try that first?

If CHANGELOG merges continue to be annoying we can re-visit more complex solutions.

alchzh, 2025-01-03:
> @alchzh, seems like the `path_ignore` option is a good first step. Can we try that first?

I guess `paths_ignore` doesn't actually work, annoyingly https://github.com/actions/runner/issues/2324