redballoonsecurity/ofrak #538: Pre-build M68k for faster Docker build times

whyitfor, 2024-12-17:
With #502, CI/CD pipeline time doubles from 20-30 to ~50 minutes.

At least 20 minutes of this time is building the M68K toolchain.
This issue would result in the toolchain being pre-built and downloaded as part of the Docker image build.
For example, a solution could be:
- Creating a new repository with a Dockerfile that builds the toolchain
- Saving a tarball of that build as a release
- Downloading that tarball in when ofrak Docker build happens

Potential issues: ensuring that the M68K toolchain can work with base images that are not `python:3.9-bookworm`.

Another option could be saving this artifact in the Docker CI/CD pipeline. One downside of this approach is that we do not provide this speedup for other OFRAK users building the Docker image.

**Which files would be affected?**
`ofrak_patch_maker/Dockerstub`, potentially others depending on how this is implemented.

**Does the proposed maintenance include non-doc string _functional_ changes to the Python code?**
No.

ANogin, 2024-12-17:
What about publishing a family of M68K toolchain docker images in a public docker image repository, than using that repository as a source of that toolchain both in CI/CD and in regular OFRAK docker builds?

ANogin, 2024-12-17:
Alternatively, building it as a .deb and having that published to public repo that could be added to apt as part of the OFRAK dependency install?