redballoonsecurity/ofrak #543: Bump Python Version for faster ARM-based docker build times

Jepson2k, 2024-12-19:
**What is the use case for the feature?**
Current arm-based docker build times are slow due to the docker build system building the wheel for the python packages z3-solver and capstone. These packages are dependencies of angr and by bumping the angr version it'll use the later version of these python packages that have pre-built binaries for ARM devices. In order to upgrade angr we need to upgrade Python.

**Does the feature contain any proprietary information about another company's intellectual property?**
No

**How would you implement this feature?**
Bumping Python, angr, and any other dependent packages.

**Are there any (reasonable) alternative approaches?**
No

**Are you interested in implementing it yourself?**
Yes

whyitfor, 2024-12-19:
What version of Python are you proposing updating to?

Jepson2k, 2024-12-29:
Current thought process:
- The newer version of angr (which uses newer versions of other packages that come with pre-build binaries for arm) requires 3.10+
- 3.11 is standard in debian bookworm and provides significant python execution speed up.
- Binwalk doesn't directly support python 3.12.8. 

After initial testing it appears that upgrading to python 3.11 requires minimal changes to maintain 100% testing coverage.