redballoonsecurity/ofrak #547: Fix ofrak-dev, ofrak-ghidra, ofrak-tutorial entrypoints

Jepson2k, 2024-12-27:
**What is the problem? (Here is where you provide a complete Traceback.)**
The addition of ofrak licensing broke the entrypoint commands for ofrak-dev, ofrak-ghidra, and ofrak-tutorial.

**Please provide some information about your environment.**
At minimum we would like the following information on your platform and Python environment:
  - Platform: redballoonsecurity/ofrak/dev, .../ghidra, .../tutorial

**If you've discovered it, what is the root cause of the problem?**
Ofrak license needs to be run first

**How often does the issue happen?**
Everytime using those configs.

**What are the steps to reproduce the issue?**
Ideally, give us a short script that reproduces the issue.
Build those images, start those images.

**How would you implement this fix?**
Add `ofrak license --community --i-agree` command to the entrypoint commands.

**Are there any (reasonable) alternative approaches?**
No

**Are you interested in implementing it yourself?**
Yes

ANogin, 2024-12-27:
@Jepson2k , we should not be submitting either `--community` or `--i-agree` options on behalf of the user - we need the user to possess the appropriate license (e.g. either explicitly confirming their use qualifies for the Community License or purchasing an appropriate license authorizing Commercial Use), and explicitly agreeing to the terms of the license. Off the top of my head, one possibility would be to provide a web UI for this, another is to enable the user to provide the license inputs at the docker build time.

Jepson2k, 2025-01-03:
@ANogin Took a different approach:
- Upon the first start of a docker container it checks for the existence of the ofrak.license file.
  - If it finds it, it attaches it to the container and proceeds as normal.
  - If it does not find it, it first brings up the freshly build container running `ofrak license` command. Prompts user for input and if the user selects the license and agrees to the terms it copies the license file out of the container and then proceeds as normal.
 
Thoughts?