redballoonsecurity/ofrak #585: Dockerfile syntax 1.3 causes build errors.

whyitfor, 2025-05-07:
**What is the problem? (Here is where you provide a complete Traceback.)**
Several users have encountered build errors due to seemingly using a newer version of of Docker syntax.

**Please provide some information about your environment.**
At minimum we would like the following information on your platform and Python environment:
  - Platform:
    - If running in one of the OFRAK Docker images, say which image it is (for example, `redballoonsecurity/ofrak/ghidra`).
    - If not running in an OFRAK Docker image, please provide the output of `python3 -m platform`.
  - Python environment: Please provide the output of `python3 -m pip freeze`.

**If you've discovered it, what is the root cause of the problem?**
Currently `build_image.py` specifies: `# syntax = docker/dockerfile:1.3`.

When the build is done on a newer version of Docker, the following type of error is hit:

```
base.Dockerfile:1
--------------------
   1 | >>> # syntax = docker/dockerfile:1.3
   2 |     ### ofrak/frontend/Dockerstage
   3 |     FROM node:latest AS svelte
--------------------
ERROR: failed to solve: unlazy requires an applier
Error running command: 'docker build --build-arg BUILDKIT_INLINE_CACHE=1 --cache-from redballoonsecurity/ofrak/scr-base:master -t redballoonsecurity/ofrak/scr-base:2fa31518 -t redballoonsecurity/ofrak/scr-base:latest -f base.Dockerfile .'
```

**How often does the issue happen?**

**What are the steps to reproduce the issue?**
This has been observed on Docker version 27.0.3, with 0.15 buildkit, and Docker version 28.1.1.

Workarounds have included:
- Deleting this line from dockerfile
- Running `docker pull docker/dockerfile:1.3` before rebuilding

**How would you implement this fix?**
There seem like a couple options:

1. Include instructions to run `docker pull docker/dockerfile:1.3` if we want to maintain reproducibility, so users can bypass this error when it happens.
2. Make the syntax line configurable.
3. Use the recommended syntax of 1: https://docs.docker.com/build/concepts/dockerfile/#dockerfile-syntax. This will always use toe most recent stable syntax.

**Are there any (reasonable) alternative approaches?**
Possibly.

**Are you interested in implementing it yourself?**
Sure, once consensus is reached.

whyitfor, 2025-05-07:
Seems like this is duplicating #578 -- I think this issue provides a few more fixes suggestions.

rbs-jacob, 2025-05-07:
We use this special syntax because it's required for BuildKit, and we use BuildKit so we can use secrets for Binary Ninja, right?

Just making sure I am correct about why this is even here in the first place.

whyitfor, 2025-05-07:
Secrets is one use case. IIRC buildkit also has some nice properties w.r.t. caching layers more efficiently

rbs-jacob, 2025-05-07:
> Secrets is one use case. IIRC buildkit also has some nice properties w.r.t. caching layers more efficiently

Sure, but it runs by default in newer Docker versions with no need to specify the syntax string in the Dockerfile to use it, right? So the only reason for the syntax line is to use secrets with Binary Ninja?